divyanshupundir/mavlink-kotlin #13: Get Bitmask flags from appropriate fields

yaroslavkulinich, 2022-11-03:
MavEnumValue.fromMask() is used to create a MavEnumValue from a few bitmask enum values.
But it seems that there is no API to get enabled bitmask enum values from a field (analog of EnumValue.flagsEnabled() in DroneFleet library). 

In the README you show how to create it:

```
val values = listOf(MavModeFlag.MAV_MODE_FLAG_AUTO_ENABLED, MavModeFlag.MAV_MODE_FLAG_SAFETY_ARMED)
val baseMode: MavModeFlag = MavEnumValue.fromMask(values)
```

But do not show how to get it:
```
val baseMode: MavModeFlag = heartbeat.baseMode
val values: Collection<MavModeFlag> = MavModeFlag.someMissingMagicFunction(baseMode) <---- ???
```

yaroslavkulinich, 2022-11-03:
Would be great to have some kind of MavBitmaskValue that extends MavEnumValue and have one additional field called 'flags'.
So to get all enabled flags will be like `val enabledFlags: List<MavModeFlag> = heartbeat.baseMode.flags` where 'baseMode' is MavBitmaskValue.

divyanshu1234, 2022-11-03:
This actually sounds interesting. I didn't use to use bitmasks too much before so didn't need this functionality. I'll try to do some research and get back in a couple of days.

But at glance, it looks like there needs to be some additional code generation to deserialize the flags in the respective enum companion objects.

The `MavBitmaskValue` does look like a good idea. We can replace `MavEnumValue` with it where it should've been a bitmask instead. This will also make the `MavEnumValue` class a bit smaller.

yaroslavkulinich, 2022-11-05:
@divyanshu1234 thanks. It looks absolutely ok. 

I have big hopes for your library migrating from reflection-based approach. Look forward to Multiplatform support for the generated model and frame encoding-decoding logic.

divyanshu1234, 2022-11-09:
@yaroslavkulinich, this has been implemented in v0.10.1-alpha.

You can download the packages and test in your project. Please let me know in case you find a bug. I'll try to push a stable version in a couple of days.

yaroslavkulinich, 2022-11-10:
@divyanshu1234 Thank you for such a fast implementation of this feature!