divyanshupundir/mavlink-kotlin #15: MISSION_REQUEST ( #40 ) message wrong parameter order serialization/deserialization

yaroslavkulinich, 2022-11-11:
After migration from Dronefleet library to mavlink-kotlin I started testing my app's functionality to check if everything works.
But I faced a problem uploading mission to Autopilot (SITL). The reason is MISSION_REQUEST ( #40 ) message I receive from Autopilot. It seems wrongly deserialized, so I received the wrong seq parameter and can't proceed with the process of mission upload.

I checked the same mission upload process with Dronefleet implementation - everything works as expected. 

Then I started to check the deserialization phase on both Dronefleet and mavlink-kotlin implementations and noticed a pretty obvious thing - the order of byte reading in both libs is different - that's why I receive MissionRequest message with shuffled values in parameters. 
I expect to receive :
`MissionRequest(targetSystem=255, targetComponent=1, seq=0,  missionType = 0)` 
but receive :
`MissionRequest(targetSystem=0, targetComponent=255, seq=1,  missionType = 0)`

Bytes payload for this message is  [0, 0, -1, 1]. 
Deserializarion results: 
Dronefleet (correct) ->
`MissionRequest{targetSystem=255, targetComponent=1, seq=0, missionType=EnumValue{value=0, entry=MAV_MISSION_TYPE_MISSION}}`
Mavlink-kotlin (shuffled params) -> 
`MissionRequest(targetSystem=0, targetComponent=255, seq=1, missionType=MavEnumValue(entry=MAV_MISSION_TYPE_MISSION, value=0))`

If you serialize and deserialize MISSION_REQUEST with mavlink-kotlin - everything will be ok. But only for mavlink-kotlin. It's obvious because the processing order in generated logic for serialization and deserialization is the same.

One more thing. Maybe I had to start with it. To deserialize the MISSION_REQUEST message with mavlink-kotlin I skipped validateCrc() step. Because it was failing.

If problem with the params processing order then it appears to be an issue in all similar messages.

Version of mavlink-kotlin: 0.9.0.

divyanshu1234, 2022-11-11:
Okay, that looks like a problem with parameter ordering. I'll have a look into it and post a fix soon. Thanks again.

divyanshu1234, 2022-11-11:
From what I have figured out is that it is a problem with messages that have extensions and are deprecated as well. It is a problem with the XML reading part.

yaroslavkulinich, 2022-11-11:
Thank you for response!
This library is perfect fit for my project in terms of performance and convenience.

divyanshu1234, 2022-11-11:
I have fixed the problem in version 0.10.2-alpha. I have done some tests and it seems good. Please test it yourself too and I'll create a stable by tomorrow.

Thanks for the kind words!

yaroslavkulinich, 2022-11-12:
Just tested. Everything ok! That's perfect! Thank you!

divyanshu1234, 2022-11-12:
Awesome!

closed 2022-11-12 by divyanshu1234

divyanshu1234, 2022-11-12:
The latest changes are in v0.11.0