divyanshupundir/mavlink-kotlin #28: How to Decode byte array?

JayParikh20, 2023-05-31:
I'm getting mavlink messages in byte array from flight controller on my android app, how do I use this library to decode it?

divyanshupundir, 2023-06-01:
Currently, the library only supports TCP server and client endpoints. How we achieve multiple connections in our app is by using frontend connections for UDP, Serial, Bluetooth, etc. and forwarding the bytes to an internal TCP server to which mavlink-kotlin connects.

I understand that this is somewhat of a longer method, and I have plans on restructuring the base `mavlink-kotlin` module to provide in-built support for the connections. For example, `endpoint-tcp`, `endpoint-udp`, `endpoint-serial-android`, etc. I am just not able to find time for doing that.

JayParikh20, 2023-06-01:
@divyanshupundir is there any way to use it without any connection, like a function thats takes byte array and returns Mavlink output?

divyanshupundir, 2023-06-01:
With reference to the [MAVLink Serialization Scheme](https://mavlink.io/en/guide/serialization.html), if you've extracted the `payload` bytes from the MAVLink Frame and you know what type it is, then you can do this:

```kotlin
val m = MessageType.classMetadata.deserializer.deserialize(payloadBytes)
```

For example:
```kotlin
val m = Heartbeat.classMetadata.deserializer.deserialize(payloadBytes)
```

JayParikh20, 2023-06-01:
This is perfect, Thank you!

closed 2023-06-01 by JayParikh20

divyanshupundir, 2023-06-01:
Awesome!

divyanshupundir, 2023-06-04:
@JayParikh20 as of `v0.19.0` you can deserialize a message like this:

```kotlin
val m = Heartbeat.deserialize(payloadBytes)
```

JayParikh20, 2023-06-05:
> @JayParikh20 as of `v0.19.0` you can deserialize a message like this:
> 
> ```kotlin
> val m = Heartbeat.deserialize(payloadBytes)
> ```

Oh this is even better, thanks for the library. Its really helpful!