divyanshupundir/mavlink-kotlin #33: Consider Korlibs for multiplatform

yaroslavkulinich, 2023-07-04:
Hello again! 
I saw you made some progress using Okio to support multiplatform use cases but got poor performance results.
Consider Korlibs set of libraries for developing multiplatform features. They have pretty big amount of elementary functionality chunked by small multiplatform libraries. For example bytes, arrays, buffers utilities gathered in [Kmem ](https://docs.korge.org/kmem/)  library. They have some gaps in documentation but really broad number of utilities. Their main page with all libs described is [here](https://docs.korge.org/) .
I don't know about pros and cons of specific features performance comparing to other multiplatform libs like Okio, but I think it can be a helpful for your work at least to know about it.
Thanks.

divyanshupundir, 2023-07-05:
Hello @yaroslavkulinich.

Thank for sharing this, it looks pretty good. I did not know about this. Kmem should fit the requirement well. Okio is pretty good and I am still using for the byte stream reading and writing. But mavlink-kotlin requires a lot of encoding from and decoding to ByteArray where I feel Okio is not optimised.

For the time being, I abstracted the encoding and decoding with [MavDataDecoder](https://github.com/divyanshupundir/mavlink-kotlin/blob/main/serialization/src/main/kotlin/com/divpundir/mavlink/serialization/MavDataDecoder.kt) and  [MavDataEncoder](https://github.com/divyanshupundir/mavlink-kotlin/blob/main/serialization/src/main/kotlin/com/divpundir/mavlink/serialization/MavDataEncoder.kt). I was planning to use the expect-actual kotlin feature to implement if for multiplatform in the future.

For the time being I am working on [this](https://github.com/divyanshupundir/mavlink-kotlin/tree/restructure) branch to restructure the library to support more connections than just TCP. I'm planning to get this thing done first for the JVM and release the v1 of the library.

In a few weeks/months I'll be starting the work on multiplatform.

yaroslavkulinich, 2023-07-05:
It is an excellent plan.
My other 5 cents - please make the data class MavRawFrame not internal but public, because it blocks developers from creating their own implementations of low-level connection logic. For example, I have implemented my own TCP and UDP connections based on different stacks (Java sockets and Ktor sockets) with custom filtering logic to prevent redundant deserialization of packets that my system never uses (to decrease processing load). But I'm forced to use an old version of the library to keep it working because MavRawFrame is public there and internal in a newer versions.

divyanshupundir, 2023-07-05:
Okay great, I will make the MavRawFrame and MavRawFrameReader public when I create a PR. I'll also try to add documentation for that.

If you want, you can track the related changes in the branch and let me know if there's anything else to be changed. I'll try to finish the PR either by this weekend or the next one.

yaroslavkulinich, 2023-07-05:
Great! 
Thank you again for the library.

divyanshupundir, 2023-07-05:
Thanks for the support and ideas.

divyanshupundir, 2023-07-09:
@yaroslavkulinich please refer #34. I have tested most things and will try to create a release today.

yaroslavkulinich, 2023-07-09:
> @yaroslavkulinich please refer #34. I have tested most things and will try to create a release today.

Thank you!

closed 2023-07-10 by divyanshupundir