divyanshupundir/mavlink-kotlin #48: Ardupilotmega MavCmd commands using CommandLong class

yaroslavkulinich, 2023-09-15:
The question is how I can send ardupilotmega (or any other 3rd-party dialect) MavCmd command using the CommandLong message.
I see the CommandLong class accepts only MavCmd from the common package.

Or maybe I'm missing something? 

Thanks

yaroslavkulinich, 2023-09-15:
Okay. I handled it.
The trick is to use MavEnumValue.from<xyz.urbanmatrix.mavlink.definitions.common.MavCmd>(xyz.urbanmatrix.mavlink.definitions.ardupilotmega.MavCmd.MAV_CMD_SCRIPTING.value)

closed 2023-09-15 by yaroslavkulinich

divyanshupundir, 2023-09-15:
Yes that's how we need to do it. I wanted to make it a bit better, but it is rarely used.

To make it a bit more clean you can use import alias of ApmCmd for Ardupilotmega commands.