divyanshupundir/mavlink-kotlin #53: Multiplatform definitions

yaroslavkulinich, 2023-11-04:
Consider to make definitions module (message and enum classes) as Kotlin Multiplatform module, even with one target (JVM).
It will make life of KMP developers much easier. Because currently I can't use this library in common sourceSet - I need to make expect-actual procedure for every message and field I'm going to use. It works, but from maintenance and further development perspective it's really time-consuming operation.

divyanshupundir, 2023-11-05:
Hello @yaroslavkulinich.

Kotlin multi-platform has recently become stable. I had been waiting for this.

Please give me time till next weekend. I'm going to start the multi-platform setup and work in a few days.

yaroslavkulinich, 2023-11-06:
@divyanshupundir thank you! Perfect news!

divyanshupundir, 2023-11-12:
Hello @yaroslavkulinich, I have been trying to shift the codebase to multiplatform. I am currently working on the serialization module. You had told me about the Kmem library in this issue #33. Looks like they have made some changes in the project. I am unable to find the library. Can you please let me know how to get access to it, or some other library that could fit the usecase?

yaroslavkulinich, 2023-11-12:
Hello @divyanshupundir.
It seems like they moved this functionality to foundation library.
Valid link with description here: https://docs.korge.org/memory/#

divyanshupundir, 2023-11-13:
Thanks. For now, I am using simple ByteArray and bit manipulation for the data encoding and decoding for non-JVM targets. It is working for now. I will look into optimization and other libraries once I have mavlink-kotlin running on the different platforms.

divyanshupundir, 2023-11-25:
@yaroslavkulinich, I have ported the library to multiplatform and version `1.2.0-alpha.5` should be available on maven in sometime.

I had to make several internal changes for this, but I've tried my best to keep the API changes to a minimum. Since it's a big migration, I will be testing the library for a few days before bumping it to stable.

Currently, the TCP and UDP connections are available only on JVM.

Contributions for adding the missing functionalities are welcome.

yaroslavkulinich, 2023-11-29:
@divyanshupundir Thank you! You do a great job!
Message and Enum classes in common sourceSet would be enough to remove 90% of the headache.
Transport is not an issue.

divyanshupundir, 2023-11-30:
Thanks a lot for the kind words!

Please do let me know if the serialization and deserialization of the messages are working fine. Currently, I have the means to test it only on Linux and JVM. If it works fine, I'll create a release this weekend or next week.

yaroslavkulinich, 2023-12-13:
I was testing it for a few days and seems it works good.
Good job 👍

divyanshupundir, 2023-12-14:
Awesome, I'll close this soon.

divyanshupundir, 2023-12-15:
mavlink-kotlin v1.2.0 has been released with multiplatform support. Documentation will be update soon.

closed 2023-12-15 by divyanshupundir