divyanshupundir/mavlink-kotlin #57: UdpServerMavConnection connect(), close() blocks when no Mavlink message received

yaroslavkulinich, 2023-12-06:
The connect() function, when used with UdpServerMavConnection(), currently blocks if there is no active Mavlink feed. Additionally, attempting to close this connection using close() also leads to blocking. This behavior might be intentional, but it poses challenges. Often, it's necessary to establish a connection before the Mavlink feed is available, or to close the connection gracefully in cases like entering an incorrect address. The current design impedes these operations due to its blocking nature.

It's important to differentiate between two aspects: the transport connection and the Mavlink component connection. A Mavlink component connection is considered active while receiving Mavlink frames from the Mavlink component. This connection can become inactive if no Mavlink frames are received for a certain period. The status of the connection at the Mavlink component level is more complex and varies based on the presence or absence of the feed from different components.

Given these complexities, I suggest focusing on tracking the status of the transport connection, rather than the Mavlink component connection status. For instance, the status could be 'Active' when the UDP server connection is started, 'Stopped' when it's stopped, and 'Failed' if the connection attempt fails (as implemented but mirroring transport connection). This approach simplifies the tracking of connection status by focusing on the underlying transport mechanism, rather than the variable state of Mavlink components.

divyanshupundir, 2023-12-07:
@yaroslavkulinich, I have partially understood the problem and will try to answer the queries based on that. If you don't find the explanations convincing, I would request you to give some code examples. There could even be bugs that I haven't seen.

mavlink-kotlin is designed to work with multiple threads and that's why it's recommended that you use the provided adapters.

All the `connect()` methods are made blocking so that they can be fired up in a background thread, and we know that the transport connection is successful when the code moves to the next line.

At the same time, you can observe the `streamState` reactive steam (transport connection state) in another thread. As for the MAVLink component connection, you can also observe the `mavMessage` reactive stream in another thread and act accordingly (maybe call `close()`) if there are no heartbeats for 30s. The library gives you that choice.

Regarding the `UdpServerMavConnection`, it follows the same rules. One clarification, it doesn't specifically wait for MAVLink messages, rather, any 32-byte ping can trigger the connection.

Given this, I highly recommend you to use the adapters, or if you want to use your own threading mechanism, please have a look at the adapter implementations.

yaroslavkulinich, 2023-12-08:
@divyanshupundir You can't normally use `close()` while `connect()` is blocked (waiting for the ping). That's because you get the exception  `java.io.IOException: The connection is already closed`. As a result Connection is still blocked. 

I was wrong in my previous statement (the exception was not printed in `catch` block, so I thought `close()` blocked):

> Additionally, attempting to close this connection using close() also leads to blocking.

divyanshupundir, 2023-12-10:
Okay. I get the issue now. Let me see how I can interrupt the blocking of the server port in case `close()` is called.

divyanshupundir, 2023-12-10:
Also, can you please confirm if the multi-platform definition PR is working well. I'll add the fix post merging that.

divyanshupundir, 2023-12-16:
@yaroslavkulinich, I have made the required changes in PR #58. The library is being published and should be available as version `1.2.1-alpha` in a couple of hours.

divyanshupundir, 2024-01-16:
@yaroslavkulinich, did you test it out?

divyanshupundir, 2024-02-06:
@yaroslavkulinich any update?

PR #58 (2024-02-20): feat: Add the ability to interrupt the opening state of connections
This is particularly helpful in the case of server connections. As the server connections are being made, they block the thread, and due to the design of the `AbstractMavConnection`, the library did not provide any way to interrupt the connection.

This PR fixes this problem.

Fixes #57
closed 2024-02-20 by divyanshupundir