divyanshupundir/mavlink-kotlin #63: FYI MAVLink XML/XSD update to allow exponentation operator in bitmask flags

hamishwillee, 2024-02-15:
FYI We're modifying the MAVLink XSD validation file in https://github.com/ArduPilot/pymavlink/pull/920 to allow a bitmask value to be declared using the Python exponentiation operator. 

This makes it easier to immediately visualize which bit is set by a flag, because you can use the syntax 
```xml
<entry value="2**15" name="BIT15" />
```
rather than
```xml
<entry value="32768" name=" BIT15" />
```
This is transparent to mavgen because the parser evaluates the number before creating the generated headers. However it may affect other the rust parser
We're not updating the XML yet to allow other parsers time to update.

Any concerns, please raise on https://github.com/ArduPilot/pymavlink/pull/920

divyanshupundir, 2024-02-15:
Hello @hamishwillee.

Thank you for the update. I'll integrate this functionality soon.

divyanshupundir, 2024-02-23:
Solved in #65

closed 2024-02-23 by divyanshupundir

hamishwillee, 2024-02-25:
Excellent!