divyanshupundir/mavlink-kotlin #66: Program blocks on s.connection.next() on Android but works fine on desktop

hiquanta, 2024-02-21:
Description
I encountered an issue with [Library Name] when performing cross-platform network communication. Specifically, when attempting to read MAVLink messages from a UDP connection, the code runs smoothly and receives data on the desktop. However, on Android, the same code blocks indefinitely in the nextRawFrame() function at the line when (peeked.readByte().toUByte()). This issue seems to occur when trying to parse the received raw frame, more precisely, when reading the frame header to determine the frame version.

Steps to Reproduce
Establish a UDP connection to [Server/Device] on both desktop and Android.
Attempt to read MAVLink messages by calling s.connection.next().
On the desktop, the program receives and processes data normally.
On Android, the program blocks in the nextRawFrame() function and cannot proceed or receive data.
Expected Behavior
I expect s.connection.next() to read and return MAVLink messages on Android as it does on the desktop, or to avoid indefinite blocking by setting a timeout when data is not available.

Actual Behavior
On the desktop, the program operates as expected, receiving and parsing MAVLink messages. However, on Android, the program indefinitely blocks when attempting to read and parse MAVLink messages, especially at the nextRawFrame() function's line when (peeked.readByte().toUByte()). The program does not timeout or exit the blocking state even when no data arrives.

Environment
Test device/emulator: [android 12]
Operating system version: [windows11]
Attempted Solutions
Confirmed that the network connection is normal, and the server/device has sent data.
Attempted to run the program in different network environments to rule out network issues.
Checked and set read operation timeouts, but the problem persists.
Additional Information
          Button(onClick = {
                println("click test!")
                CoroutineScope(Dispatchers.IO).launch {
                    try {
                        val connection = UdpServerMavConnection(20008, CommonDialect)
                        connection.connect()
                        repeat(1000) {
                            println("repeat!")
                            println(connection.next())
                        }
                        println("connection.close()!")
                        connection.close()
                    }catch (e:Exception){
                        println("error:"+e.message)

                    }

                }
            }) {
                Text("connect test")
            }
Since this issue only occurs on Android and not on the desktop, I suspect there might be specific constraints or configuration issues with the Android platform. I'm hoping to get some help on how to resolve this blocking issue on Android. Thank you!

hiquanta, 2024-02-22:
Sorry, I used it wrong. I should use adapters to get the message.

closed 2024-02-22 by hiquanta

divyanshupundir, 2024-02-22:
Awesome!

Indeed you should use the coroutines adapter. That will take care of the threading for you.