divyanshupundir/mavlink-kotlin #79: AUTOPILOT_VERSION (148) wrong field decoding order

yaroslavkulinich, 2024-05-20:
Generated AutopilotVersion class has wrong field decoding order. As result we have shuffled fields and no way to properly get some fields.
Field order in constructor is correct BTW.

[AUTOPILOT_VERSION](https://mavlink.io/en/messages/common.html#AUTOPILOT_VERSION)

divyanshupundir, 2024-05-21:
@yaroslavkulinich, the field ordering for serialisation and deserialization is done according to the [MAVLink field ordering scheme](https://mavlink.io/en/guide/serialization.html#field_reordering).

You can compare it with other MAVLink libraries, [dart_mavlink](https://github.com/nus/dart_mavlink/blob/d649b46bbc78e711378590f4f9d1b513163dd95b/lib/dialects/common.dart#L24680) for example. You can see that the ordering is same for `AUTOPILOT_VERSON`.

I had kept the ordering in the constructor according to their arrangement in the MAVLink XML definitions so that it is easy to correlate while using the class.

yaroslavkulinich, 2024-05-21:
@yaroslavkulinich okay, thanks.
I will try dig deeper with the issue. 

But without additional analysis for me it's clear that MissionPlanner correctly represents AUTOPILOT_VERSION message (described in mavlink inspector), but with mavlink-kotlin deserialization I get shuffled fields. For the same message.

I'm using ardupilot SITL. Version 4.6.0 dev.

This problem is something new for me, because I never faced it with other messages.

divyanshupundir, 2024-05-22:
@yaroslavkulinich, I have checked the message with both PX4 and Ardupilot on real flight controllers. I am getting the correct data. When we print the message using the implicit `toString()`, the fields will get printed according to their order in the constructor.Can you please provide screenshots

divyanshupundir, 2024-06-19:
Closing due to inactivity.

ZhipengJuly, 2024-07-10:
@yaroslavkulinich Hi, I got the same issue in version 1.2.6, here is the log and capture from wireshark 
```
2024-07-10 18:15:34.268 25776-25817 FlyController           com.test.sample                 D  onMessageReceived: TerrainReport(lat=0, lon=0, spacing=0, terrainHeight=0.0, currentHeight=0.0, pending=112, loaded=0)
2024-07-10 18:15:34.268 25776-25817 FlyController           com.test.sample                 D  onMessageReceived: Vibration(timeUsec=11689756109, vibrationX=0.03799032, vibrationY=0.0223004, vibrationZ=0.02042753, clipping0=0, clipping1=0, clipping2=0)
2024-07-10 18:15:34.270 25776-25817 FlyController           com.test.sample                 D  onMessageReceived: BatteryStatus(id=0, batteryFunction=MavEnumValue(entry=UNKNOWN, value=0), type=MavEnumValue(entry=UNKNOWN, value=0), temperature=32767, voltages=[947, 65535, 65535, 65535, 65535, 65535, 65535, 65535, 65535, 65535], currentBattery=24, currentConsumed=809, energyConsumed=27, batteryRemaining=75, timeRemaining=0, chargeState=MavEnumValue(entry=OK, value=1), voltagesExt=[0, 0, 0, 0], mode=MavEnumValue(entry=UNKNOWN, value=0), faultBitmask=MavBitmaskValue(flags=[], value=0))
2024-07-10 18:15:34.270 25776-25817 FlyControlActivity      com.test.sample                 D  onBatteryStatus: BatteryStatus(id=0, batteryFunction=MavEnumValue(entry=UNKNOWN, value=0), type=MavEnumValue(entry=UNKNOWN, value=0), temperature=32767, voltages=[947, 65535, 65535, 65535, 65535, 65535, 65535, 65535, 65535, 65535], currentBattery=24, currentConsumed=809, energyConsumed=27, batteryRemaining=75, timeRemaining=0, chargeState=MavEnumValue(entry=OK, value=1), voltagesExt=[0, 0, 0, 0], mode=MavEnumValue(entry=UNKNOWN, value=0), faultBitmask=MavBitmaskValue(flags=[], value=0))
2024-07-10 18:15:34.270 25776-25817 FlyController           com.test.sample                 D  onMessageReceived: Attitude(timeBootMs=11689773, roll=-2.9905872, pitch=0.4307207, yaw=0.78559434, rollspeed=-0.0032989802, pitchspeed=0.0018957397, yawspeed=4.4002663E-6)
2024-07-10 18:15:34.270 25776-25817 FlyController           com.test.sample                 D  onMessageReceived: VfrHud(airspeed=0.0, groundspeed=0.0060461196, heading=45, throttle=0, alt=0.0, climb=0.005788166)
2024-07-10 18:15:34.270 25776-25818 MavFlyControllerWRer    com.test.sample                 D  sendMessage: Heartbeat(type=MavEnumValue(entry=GCS, value=6), autopilot=MavEnumValue(entry=INVALID, value=8), baseMode=MavBitmaskValue(flags=[], value=0), customMode=0, systemStatus=MavEnumValue(entry=UNINIT, value=0), mavlinkVersion=3)
2024-07-10 18:15:34.281 25776-25818 MavFlyControllerWRer    com.test.sample                 D  sendMessage: CommandLong(targetSystem=1, targetComponent=1, command=MavEnumValue(entry=null, value=512), confirmation=0, param1=148.0, param2=0.0, param3=0.0, param4=0.0, param5=0.0, param6=0.0, param7=0.0)
2024-07-10 18:15:34.300 25776-25818 FlyController           com.test.sample                 D  onMessageReceived: Heartbeat(type=MavEnumValue(entry=GENERIC, value=0), autopilot=MavEnumValue(entry=ARDUPILOTMEGA, value=3), baseMode=MavBitmaskValue(flags=[MANUAL_INPUT_ENABLED, STABILIZE_ENABLED, CUSTOM_MODE_ENABLED], value=81), customMode=0, systemStatus=MavEnumValue(entry=CRITICAL, value=5), mavlinkVersion=3)
2024-07-10 18:15:34.300 25776-25818 FlyControlActivity      com.test.sample                 D  onHeatBeat: HeartbeatInfo(type=GENERIC, autopilot=ARDUPILOTMEGA, customModeEnabled=true, testEnable=false, autoEnable=false, guidedEnabled=false, stabilizeEnabled=true, hilEnabled=false, manualInputEnabled=true, safetyArmed=false, customModeInfo=)
2024-07-10 18:15:34.381 25776-25818 FlyController           com.test.sample                 D  onMessageReceived: CommandAck(command=MavEnumValue(entry=REQUEST_MESSAGE, value=512), result=MavEnumValue(entry=ACCEPTED, value=0), progress=0, resultParam2=0, targetSystem=255, targetComponent=1)
2024-07-10 18:15:34.381 25776-25818 FlyController           com.test.sample                 D  onMessageReceived: AutopilotVersion(capabilities=MavBitmaskValue(flags=[], value=0), flightSwVersion=0, middlewareSwVersion=0, osSwVersion=67307519, boardVersion=0, flightCustomVersion=[0, 0, 99, 4, 9, 18, 64, 87], middlewareCustomVersion=[97, 97, 99, 97, 100, 56, 56, 102], osCustomVersion=[0, 0, 0, 0, 0, 0, 0, 0], vendorId=0, productId=0, uid=64495, uid2=[97, 57, 55, 99, 99, 98, 52, 49, 55, 0, 36, 0, 23, 81, 50, 51, 53, 56])
2024-07-10 18:15:34.466 25776-25818 FlyController           com.test.sample                 D  onMessageReceived: Attitude(timeBootMs=11690023, roll=-2.9905877, pitch=0.43072635, yaw=0.7856079, rollspeed=-0.0045892815, pitchspeed=0.001149722, yawspeed=2.5050272E-4)
2024-07-10 18:15:34.466 25776-25818 FlyController           com.test.sample                 D  onMessageReceived: VfrHud(airspeed=0.0, groundspeed=0.005734841, heading=45, throttle=0, alt=0.0, climb=0.004777441)
2024-07-10 18:15:34.667 25776-25818 FlyController           com.test.sample                 D  onMessageReceived: GlobalPositionInt(timeBootMs=11690253, lat=0, lon=0, alt=0, relativeAlt=5294, vx=0, vy=0, vz=0, hdg=4501)
2024-07-10 18:15:34.667 25776-25818 GpsController           com.test.sample                 D  calcDistance: homeLatLng is null
```
[image]

yaroslavkulinich, 2024-08-18:
@ZhipengJuly @divyanshupundir 
Sorry for delay. I can confirm it.
I captured AutopilotVersion message from the same SITL using MissionPlanner MavLink Inspector and using mavlink-kotlin through debug.
[image]
[image]

It looks like some shift is present.

yaroslavkulinich, 2024-08-18:
@divyanshupundir 
Okay. I found the problem. During decoding the logic skips capabilities field because `safeDecodeBitmaskValue(8)` used -> with value 8 it always returns 0.  Shift is not happening and all next decoding phases are wrong.  So it's the matter of fixing the `MavDataDecoder.safeDecodeBitmaskValue(dataSize: Int): UInt` extension function.
[image]
[image]

yaroslavkulinich, 2024-08-18:
The problem affects only these 2 messages:
[image]

divyanshupundir, 2024-08-18:
Thanks @yaroslavkulinich and @ZhipengJuly for the detailed analysis. I finally got the problem. I will fix this and push an update soon.

divyanshupundir, 2024-08-18:
@yaroslavkulinich @ZhipengJuly I have fixed the issue in version `1.2.7-alpha`. Can you please test it out? I don't have access to my test setup.

If it works fine, I'll create the stable release in version `1.2.7`.

yaroslavkulinich, 2024-08-27:
@divyanshupundir Thank you!
Just checked - everything ok!

divyanshupundir, 2024-08-27:
Awesome! I'll create a release soon.

Thanks for the bug report!

closed 2024-08-27 by divyanshupundir