divyanshupundir/mavlink-kotlin #87: java.net.PortUnreachableException for wildcard UdpClientMavConnection

dph-jan, 2024-06-19:
I'm trying to connect to a simulator via Mavproxy udp forwarding but can't get it to work.

Setup:
- I use dronekit-sitl and mavproxy on a Mac and try to connect to the simulator from an Android tablet.
- I start the simulator using: `dronekit-sitl copter`
- I start mavproxy using: `mavproxy.py --master tcp:127.0.0.1:5760 --sitl 127.0.0.1:5501`
and then add a mavproxy output for the tablet: `output add 192.168.2.13:14550`
- To rule out general connection issues, I have also tried connecting to the simulator with the Ardupilot Mission Planner app and it automatically finds and connects to port 14550.

Problem:
When starting a UdpClientMavConnection with the wildcard ip 0.0.0.0, I get a java.net.PortUnreachableException. When using the mac's local ip instead of 0.0.0.0, I don't get an exception but the connect call never finishes and no frames are received.

Relevant code:
```
private var _connection: CoroutinesMavConnection? = null
private val scope = CoroutineScope(Dispatchers.Main)

init {
        val dialect = ArdupilotmegaDialect
        _connection = UdpClientMavConnection("0.0.0.0", 14550, dialect).asCoroutine()
        _connection
            .mavFrame
            .onEach { onMavFrame(it) }
            .launchIn(scope)
        _connection
            .streamState
            .onEach { onStreamState(it) }
            .launchIn(scope)
}

private fun onStreamState(state: StreamState) {
        when (state) {
            StreamState.Active -> {
                log("connection active")
            }

            StreamState.Inactive.Stopped -> {
                log("connection stopped")
                startConnection()
            }

            is StreamState.Inactive.Failed -> {
                log("connection failure ${state.cause.message}")
                state.cause.printStackTrace()
            }
        }
    }

private fun startConnection() {
        scope.launch {
            log("start connection $_connection")
            _connection?.connect(scope)
            log("after start connection $_connection")
        }
    }
```

So the connection always starts with StreamState.Inactive.Stopped which I assume is the expected behaviour. When trying to start the connection with "0.0.0.0", I get the java.net.PortUnreachableException. When trying to start with the mac's local ip, it reaches the log statement "start connection" but not "after start connection" and no frames are received.

divyanshupundir, 2024-06-19:
@dph-jan, please correct me if I'm wrong. You're routing MAVLink packets from your PC to your Android tablet. In that case, your PC is acting as a UDP broadcaster. You should use `UdpServerMavConnection` instead of client.

Please try this:

```kotlin
_connection = UdpServerMavConnection(14550, dialect).asCoroutine()
```

dph-jan, 2024-06-19:
That seems to have fixed it, thank you! Just for general understanding, when would you use the UdpClientMavConnection instead?

closed 2024-06-19 by dph-jan