divyanshupundir/mavlink-kotlin #88: Mavlink multicast support for telemetry data

GeorgePapageorgakis, 2024-06-20:
I read in the official MAVLink overview page [here](https://mavlink.io/en/about/overview.html):

MAVLink is a binary telemetry protocol designed for resource-constrained systems and bandwidth-constrained links. MAVLink is deployed in two major versions: v1.0 and v2.0, which is backwards-compatible (v2.0 implementations can parse and send v1.0 packets). Telemetry data streams are sent in a multicast design while protocol aspects that change the system configuration and require guaranteed delivery like the [mission protocol](https://mavlink.io/en/services/mission.html) or [parameter protocol](https://mavlink.io/en/services/parameter.html) are point-to-point with retransmission.

So, does connection-udp support multicast rx/tx? If yes, which one exactly? If no, do you plan to add this feature?

divyanshupundir, 2024-06-20:
mavlink-kotlin is a low-level MAVLink library designed primarily for endpoints. Multicast and p2p behaviour is something that the user of this library has to keep in mind while building their routing apps.

> So, does connection-udp support multicast rx/tx?
> If yes, which one exactly?
> If no, do you plan to add this feature?

connection-udp supports UDP server and client configurations. The endpoint that you connect to needs to handle how the messages are being sent to other endpoints.

GeorgePapageorgakis, 2024-06-21:
But the protocol suggests that `Telemetry data streams are sent in a multicast design`. What exactly does this mean?

divyanshupundir, 2024-06-21:
You can read [this section](https://github.com/mavlink-router/mavlink-router?tab=readme-ov-file#detailed-description-of-capabilities) of the mavlink-router repository. It gives a good idea of MAVLink endpoints and how routing happens in a MAVLink system.

GeorgePapageorgakis, 2024-06-25:
@divyanshupundir Ok I see, however, I tried using:

`UdpServerMavConnection(port, CommonDialect).asCoroutine()`

I noticed that when a client disconnects from the server, the server cannot accept a new connection. I would expect from a "Server" to be able to accept new connections even if a client is disconnected. Do you agree?

In addition, can the `server` implementation accept multiple clients?

divyanshupundir, 2024-06-25:
If you can, please share your code snippet of reconnecting. The `UdpServerMavConnection` can reconnect multiple times if the client disconnects, but it has to be done manually.

You can read the `streamState` property to know the current state of the server and call the `connect()` method once the connection stops.

You can refer to [this part](https://github.com/divyanshupundir/mavlink-kotlin/tree/main/adapters/adapter-coroutines#managing-the-state) of the documentation.

One server can only accept one client connection.

GeorgePapageorgakis, 2024-06-25:
I am testing with ArduPilot Mission Planner.
I am connecting from Mission Planner to my app and the Stream State transists from Stopped to Active. However, when I disconnect from Mission Planner the state does not change to Stopped.

Is there a Time Out timer to detect the unexpected disconnect so that the state can change to Failed or Stopped?

divyanshupundir, 2024-06-25:
The local server is not able to figure out if the remote client is out of reach. That is one of the problems of UDP connections.

You can fix this by sending a `Heartbeat` message every second. That actually is a good practice in MAVLink systems. You can launch a coroutine to do that.

Otherwise, you can use the TCP protocol. In it, the client and server come to know about the disconnection instantly. But I don't know if Mission Planner supports that.

divyanshupundir, 2024-06-25:
You can actually even implement a timeout of your own. If you don't receive messages (or a certain message) after some time, you can disconnect the connection on your own.

Not having these things in mavlink-kotlin was a design decision. This is because in my experience with MAVLink systems different users and use cases require different configuration and handling of the connection.

Also, adding these things by the user according to their needs is trivial, thanks to the power of Kotlin.

I would recommend you read [this blog series](https://www.divpundir.com/blog/advanced-mavlink-kotlin-part-1).

GeorgePapageorgakis, 2024-06-25:
Thanks for the clarification.
In addition, I tried to get the mavlink bytes from a multicast. I get the message bytes but I cannot find a way to transform the bytes using your library. From what I see the transformation of bytes is blended in the MavConnection. I think I should use `companion.deserialize(rawFrame.payload)` in nextFrame right?

divyanshupundir, 2024-06-25:
A MAVLink frame looks like this:

[image]
(Source: mavlink.io)

In mavlink-kotlin's design, the deserialization of the static parts of a MAVLink frame happens in the connections module. Whereas, the generated message definitions contain their own serialization and deserialization logic, specifically to deserialize or serialize the *payload* part of the frame.

So, to deserialize the frame, you need to look at the connections module implementation, and to deserialize only the *payload* part, you need to look at the message definitions data classes.

divyanshupundir, 2024-06-26:
I hope the issue has been addressed. I will close this thread for now. Thank you.

closed 2024-06-26 by divyanshupundir

GeorgePapageorgakis, 2024-06-26:
Yes thanks for helping out.

I had one more question regarding targeting and creation of a Track/Point.
In the context of having a drone to detect a real world object, say a person in the field, a vehicled or whatever, and create a visual point/track; how is this information being communicated by the drone via mavlink?

I mean, what dialect/messages do I need to receive from the mavlink network to fulfill the aforementioned scenario? Are there out of the box dialects/messages that do this or do I need a custom dialect/messages?

Thank you very much!

divyanshupundir, 2024-06-26:
> I mean, what dialect/messages do I need to receive from the mavlink network to fulfill the aforementioned scenario? Are there out of the box dialects/messages that do this or do I need a custom dialect/messages?

Unfortunately, I don't have experience in this area. Maybe going through the MAVLink [microservices docs](https://mavlink.io/en/services/) may help.

You can also look at [MAVSDK](https://github.com/mavlink/MAVSDK).

> Thank you very much @divyanshupundir!

I'm glad to help!

https://mavlink.io/en/services/offboard_control.html