divyanshupundir/mavlink-kotlin #94: Help with Serial Connection

dxkruse, 2024-07-17:
Does this repo support usb serial connections or only UDP/TCP?

divyanshupundir, 2024-07-18:
@dxkruse, the `connection-core` module is designed so that users can implement their own connections for any platform. The important classes for this are [AbstractMavConnection](https://github.com/divyanshupundir/mavlink-kotlin/blob/main/connections/connection-core/src/commonMain/kotlin/com/divpundir/mavlink/connection/AbstractMavConnection.kt) and [BufferedMavConnection](
https://github.com/divyanshupundir/mavlink-kotlin/blob/main/connections/connection-core/src/commonMain/kotlin/com/divpundir/mavlink/connection/BufferedMavConnection.kt).

As an example, you can have  a look at the [JvmTcpClientMavConnection](https://github.com/divyanshupundir/mavlink-kotlin/blob/main/connections/connection-tcp/src/jvmMain/kotlin/com/divpundir/mavlink/connection/tcp/TcpClientMavConnection.kt) implementation.

In short, your `SerialMavConnection` will need to extend the `AbstractMavConnection` and override the `open()` method. It needs to return a `BufferedMavConnection` which will be created using your serial port.

`BufferedMavConnection` requires `BufferedSource`, `BufferedSink` and a `Closeable` callback to help close the port.

dxkruse, 2024-07-18:
Awesome, thank you for pointing me in the right direction!

divyanshupundir, 2024-07-20:
@dxkruse, should I close this issue if you were able to solve the problem?

closed 2024-07-28 by divyanshupundir

acras01, 2024-11-05:
Hello! I've written some code as you pointed. BufferedMavConnection is being created successfully, but when I'm trying to send heartbeat, I get error "com.fazecast.jSerialComm.SerialPortIOException: This port appears to have been shutdown or disconnected."

`class UsbMavConnection : AbstractMavConnection() {

    override fun open(): MavConnection {
        val ports = SerialPort.getCommPorts().onEach { println(it.descriptivePortName) }

        val serialPort = ports.firstOrNull { it.descriptivePortName.contains(DEVICE_NAME) }
            ?.apply {
                setComPortParameters(BAUD_RATE, 8, SerialPort.ONE_STOP_BIT, SerialPort.NO_PARITY)
                openPort()
            }

        val port = serialPort ?: throw IllegalStateException("Unable to find or open port for $DEVICE_NAME")

        val source = port.inputStream.source().buffer()
        val sink = port.outputStream.sink().buffer()

        return BufferedMavConnection(
            source = source,
            sink = sink,
            resource = { close() },
            dialect = ArdupilotmegaDialect
        ).also { println(it) }
    }
}`

divyanshupundir, 2024-11-08:
Hi @acras01, can you share the code snippet of how you are using this class?

divyanshupundir, 2024-11-13:
That's great!

nghinguyengr, 2025-05-07:
Hi @acras01 , could you please give me on how to configure the Android project to use the SerialPort?

acras01, 2025-05-07:
Sorry, I used it in desktop project

nghinguyengr, 2025-05-07:
@acras01 Wow, you're using this repo on desktop.