divyanshupundir/mavlink-kotlin #108: Can't upload missions

acras01, 2024-11-27:
As described at mavlink.io

`Mission Types [​](https://mavlink.io/en/services/mission.html#mission_types)
MAVLink 2 supports three types of "missions": flight plans, geofences and rally/safe points. The protocol uses the same sequence of operations for all types (albeit with different types of [Mission Items](https://mavlink.io/en/services/mission.html#mavlink_commands)). The mission types must be stored and handled separately/independently.

Mission protocol messages include the type of associated mission in the mission_type field (a MAVLink 2 message extension). The field takes one of the [MAV_MISSION_TYPE](https://mavlink.io/en/messages/common.html#MAV_MISSION_TYPE) enum values: [MAV_MISSION_TYPE_MISSION](https://mavlink.io/en/messages/common.html#MAV_MISSION_TYPE_MISSION), [MAV_MISSION_TYPE_FENCE](https://mavlink.io/en/messages/common.html#MAV_MISSION_TYPE_FENCE), [MAV_MISSION_TYPE_RALLY](https://mavlink.io/en/messages/common.html#MAV_MISSION_TYPE_RALLY).

NOTE

MAVLink 1 supports only "regular" flight-plan missions (this is implied/not explicitly set).`

Now in lib we have different namings, so I get errors while uploading
[image]

closed 2024-11-27 by acras01

divyanshupundir, 2024-11-28:
Hi @acras01. Can you please share your code snippet where you are getting the errors?

acras01, 2024-11-28:
@divyanshupundir sorry. the problem was in using deprecated MissionItem, not MissionItemInt

divyanshupundir, 2024-11-28:
Got it.