divyanshupundir/mavlink-kotlin #124: Fly Controller (PX4) did not receive heartbeat.

nghinguyengr, 2025-05-05:
I'm following the way that QGC sends the heartbeat. The logic in QGC works well because I see the QGC's heartbeat when I run "mavlink status" on the Fly Controller, but not with my code.

Here is the QGC's logic at MultiVehicleManager. 
`
void MultiVehicleManager::_sendGCSHeartbeat(void)
{
    LinkManager*                    linkManager = qgcApp()->toolbox()->linkManager();
    QList<SharedLinkInterfacePtr>   sharedLinks = linkManager->links();

    // Send a heartbeat out on each link
    for (int i=0; i<sharedLinks.count(); i++) {
        LinkInterface* link = sharedLinks[i].get();
        auto linkConfiguration = link->linkConfiguration();
        if (link->isConnected() && linkConfiguration && !linkConfiguration->isHighLatency()) {
            mavlink_message_t message;
            mavlink_msg_heartbeat_pack_chan(_mavlinkProtocol->getSystemId(),
                                            _mavlinkProtocol->getComponentId(),
                                            link->mavlinkChannel(),
                                            &message,
                                            MAV_TYPE_GCS,            // MAV_TYPE
                                            MAV_AUTOPILOT_INVALID,   // MAV_AUTOPILOT
                                            MAV_MODE_MANUAL_ARMED,   // MAV_MODE
                                            0,                       // custom mode
                                            MAV_STATE_ACTIVE);       // MAV_STATE

            uint8_t buffer[MAVLINK_MAX_PACKET_LEN];
            int len = mavlink_msg_to_send_buffer(buffer, &message);
            link->writeBytesThreadSafe((const char*)buffer, len);
        }
    }
`

Here is my code with mavlink-kotlin. 
FYI: My app can receive the messages from Fly Controller normally.

`
lifecycleScope.launch {
            _connection = UdpServerMavConnection(19856, CommonDialect).asCoroutine()

            // Send a heartbeat every second in a new coroutine
            launch {
                val heartbeat = Heartbeat(
                    type = MavType.GCS.wrap(),
                    autopilot = MavAutopilot.INVALID.wrap(),
                    baseMode = listOf(
                        MavModeFlag.SAFETY_ARMED,
                        MavModeFlag.MANUAL_INPUT_ENABLED,
                    ).wrap(),
                    customMode = 0u,
                    systemStatus = MavState.ACTIVE.wrap(),
                    mavlinkVersion = 3u
                )

                while (true) {
                    delay(1000)
                    val isAlive = _connection.trySendUnsignedV2(255u, 190u, heartbeat)
                    println("Is alive: $isAlive")
                }
            }

            // Read the FCU armed state from its heartbeat in a new coroutine
            launch {
                _connection
                    .mavFrame
                    .filter { it.message is Heartbeat }
                    .collect {
                        val heartbeat = it.message as Heartbeat
                        println("System-Component ID: ${it.systemId} - ${it.componentId}")
                        println("Type/Autopilot: ${heartbeat.type} ${heartbeat.autopilot}")
                    }
            }
}
`

divyanshupundir, 2025-05-05:
Hi @nghinguyengr. You will need to call `_connection.connect()` before reading or sending data. I suggest that you create the connection object (`UdpServerMavConnection(19856, CommonDialect).asCoroutine()`) outside of the coroutine scope and handle the connection and reconnection logic in a separate coroutine.

nghinguyengr, 2025-05-05:
Thank you for replying. Sorry for the lack of code.

I already have had the connection logic.

Here's my code for connection and request param list from the vehicle. Unfortunately, it does not work.
package com.gremsy.myapplication

```
import android.Manifest
import android.os.Bundle
import android.util.Log
import androidx.activity.ComponentActivity
import androidx.activity.compose.setContent
import androidx.activity.enableEdgeToEdge
import androidx.compose.foundation.layout.Column
import androidx.compose.foundation.layout.fillMaxSize
import androidx.compose.foundation.layout.padding
import androidx.compose.material3.Button
import androidx.compose.material3.Scaffold
import androidx.compose.material3.Text
import androidx.compose.runtime.Composable
import androidx.compose.ui.Modifier
import androidx.core.app.ActivityCompat
import androidx.lifecycle.lifecycleScope
import com.divpundir.mavlink.adapters.coroutines.CoroutinesMavConnection
import com.divpundir.mavlink.adapters.coroutines.asCoroutine
import com.divpundir.mavlink.adapters.coroutines.tryConnect
import com.divpundir.mavlink.adapters.coroutines.trySendUnsignedV2
import com.divpundir.mavlink.api.contains
import com.divpundir.mavlink.api.wrap
import com.divpundir.mavlink.connection.StreamState
import com.divpundir.mavlink.connection.udp.UdpServerMavConnection
import com.divpundir.mavlink.definitions.common.CommonDialect
import com.divpundir.mavlink.definitions.common.ParamRequestList
import com.divpundir.mavlink.definitions.common.ParamValue
import com.divpundir.mavlink.definitions.minimal.*
import com.gremsy.myapplication.ui.theme.MyApplicationTheme
import io.mavsdk.System
import io.mavsdk.mavsdkserver.MavsdkServer
import io.mavsdk.mission.Mission
import io.mavsdk.mission.Mission.MissionItem.VehicleAction
import io.mavsdk.param.Param
import io.mavsdk.telemetry.Telemetry
import io.reactivex.CompletableObserver
import io.reactivex.CompletableSource
import io.reactivex.disposables.Disposable
import kotlinx.coroutines.CoroutineScope
import kotlinx.coroutines.Dispatchers
import kotlinx.coroutines.delay
import kotlinx.coroutines.flow.collectLatest
import kotlinx.coroutines.flow.filter
import kotlinx.coroutines.flow.filterIsInstance
import kotlinx.coroutines.flow.map
import kotlinx.coroutines.launch
import java.util.concurrent.CountDownLatch

class MainActivity : ComponentActivity() {
    private lateinit var _connection: CoroutinesMavConnection

    override fun onCreate(savedInstanceState: Bundle?) {
        super.onCreate(savedInstanceState)
        enableEdgeToEdge()
        ActivityCompat.requestPermissions(
            this,
            arrayOf(
                Manifest.permission.INTERNET,
                Manifest.permission.ACCESS_NETWORK_STATE,
                Manifest.permission.WRITE_EXTERNAL_STORAGE
            ),
            1
[... 109 lines cut ...]
    onClick: () -> Unit,
    onClick2: () -> Unit,
    name: String,
    modifier: Modifier = Modifier
) {
    Column {
        Text(
            text = "Hello $name!",
            modifier = modifier
        )
        Button(onClick = onClick) {
            Text(
                text = "Connect",
                modifier = modifier
            )
        }
        Button(onClick = onClick2) {
            Text(
                text = "Request param",
                modifier = modifier
            )
        }
    }
}
```

divyanshupundir, 2025-05-05:
@nghinguyengr the required implementation looks fine to me. Can you check the heartbeat using another MAVLink observability tool? Maybe QGC only reads the heartbeat messages from specific sources.

I also think that since PX4 requires GCS heartbeats to start sending data itself (I checked this over a year ago; things might have changed), the heartbeats might be reaching PX4.

nghinguyengr, 2025-05-05:
> [@nghinguyengr](https://github.com/nghinguyengr) the required implementation looks fine to me. Can you check the heartbeat using another MAVLink observability tool? Maybe QGC only reads the heartbeat messages from specific sources.

When I run "mavlink status" on QGC, it will show the whole response from the vehicle. So I do not think heartbeat messages will be excluded.

> I also think that since PX4 requires GCS heartbeats to start sending data itself (I checked this over a year ago; things might have changed), the heartbeats might be reaching PX4.

Yes @divyanshupundir , you're absolutely right.

nghinguyengr, 2025-05-05:
> Can you check the heartbeat using another MAVLink observability tool?

Let me try.

nghinguyengr, 2025-05-05:
> > Can you check the heartbeat using another MAVLink observability tool?
> 
> Let me try.

Hi @divyanshupundir , I'm using https://packetsender.com/ to check the heartbeat.

I cannot connect to the port and did not send any message, but the Sender app receives strange messages.

TIME	From IP	From Port	To IP	To Port	Method	Error	ASCII	Hex
2025-05-05 17:54:30.797	192.168.40.50	39081	You	8080	UDP		\00\00\00\00\00\00\00\00\00\00\00\00\00\00\00\00\00\00\00\00\00\00\00\00\00\00\00\00\00\00\00\00	00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 	
2025-05-05 17:54:30.623	192.168.40.50	44124	You	8080	UDP		\00\00\00\00\00\00\00\00\00\00\00\00\00\00\00\00\00\00\00\00\00\00\00\00\00\00\00\00\00\00\00\00	00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 	
2025-05-05 17:54:30.581	192.168.40.50	45764	You	8080	UDP		\00\00\00\00\00\00\00\00\00\00\00\00\00\00\00\00\00\00\00\00\00\00\00\00\00\00\00\00\00\00\00\00	00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 	

Here's my code.

```
lifecycleScope.launch {
            _connection = UdpClientMavConnection("192.168.20.20", 8080, CommonDialect).asCoroutine()
            // Block the current coroutine till the connection is made for the first time
            while (!_connection.tryConnect(this@launch)) {
                println("Try connect")
                delay(1000)
            }
}
```

Do you think that anything is missing in JvmUdpServerMavConnection? 

```
private class JvmUdpServerMavConnection(
    private val port: Int,
    private val dialect: MavDialect
) : AbstractMavConnection(), UdpServerMavConnection {

    @Volatile
    private var channel: DatagramChannel? = null

    @Throws(IOException::class)
    override fun open(): MavConnection {
        val ch = DatagramChannel.open().apply {
            configureBlocking(true)
            bind(InetSocketAddress(port))
        }

        channel = ch

        ch.connect(ch.receive(ByteBuffer.allocate(32)))

        return BufferedMavConnection(
            ByteChannelSource(ch).buffer(),
            ByteChannelSink(ch).buffer(),
            ch,
            dialect
        )
    }

    @Throws(IOException::class)
    override fun interruptOpen() {
        super.interruptOpen()
        channel?.close()
    }
}
```

divyanshupundir, 2025-05-05:
That is strange. We used to use mavlink-kotlin at several places in our stack and did not notice this issue. I don't have my MAVLink testing setup now since I haven't been working in the drone industry for some time.

I used to debug data streams between two nodes by adding a bridge node that would simply relay the messages between the 2 nodes, but would also print the required message from the concerned nodes.

You can use [gomavlib](https://github.com/bluenviron/gomavlib) for this. You can even try out [mavlink-router](https://github.com/mavlink-router/mavlink-router)

nghinguyengr, 2025-05-06:
> That is strange. We used to use mavlink-kotlin at several places in our stack and did not notice this issue. I don't have my MAVLink testing setup now since I haven't been working in the drone [...]

Let me check several cases if any issues:

- Sending/receiving heartbeat messages: Whether mismatch between serialization/deserialization.
- Problem sending data.

nghinguyengr, 2025-05-06:
> You can use [gomavlib](https://github.com/bluenviron/gomavlib) for this. You can even try out [mavlink-router](https://github.com/mavlink-router/mavlink-router)

The mavlink-kotlin is perfect. I'll try to find the issue. I hope I can contribute to this repo.

nghinguyengr, 2025-05-06:
After testing some things, I think the issue is related to connections. 

**UDP Client**

I've changed the JvmUdpClientMavConnection, the connection sends heartbeat messages correctly.

JvmUdpClientMavConnection:

```
    @Throws(IOException::class)
    override fun open(): MavConnection {

        try {
            val ch = DatagramChannel.open().apply {
                configureBlocking(true)
                connect(address)
            }
            channel = ch
            return BufferedMavConnection(
                ByteChannelSource(ch).buffer(),
                ByteChannelSink(ch).buffer(),
                ch,
                dialect
            )
        } catch (e: IOException) {
            println("Exception: $e")
            throw e
        }

    }
```

Usage:

```
        lifecycleScope.launch {
            _connection =
                _UdpClientMavConnection("192.168.20.20", 8080, CommonDialect).asCoroutine()
            try {
                _connection.connect(this@launch)

            } catch (e: Exception) {
                println("Exception: $e")
            }
//             Send a heartbeat every second in a new coroutine
            launch {
                val heartbeat = Heartbeat(
                    type = MavType.GCS.wrap(),
                    autopilot = MavAutopilot.INVALID.wrap(),
                    baseMode = listOf(
                        MavModeFlag.SAFETY_ARMED,
                        MavModeFlag.MANUAL_INPUT_ENABLED,
                    ).wrap(),
                    customMode = 0u,
                    systemStatus = MavState.ACTIVE.wrap(),
                    mavlinkVersion = 3u
                )

                while (true) {
                    delay(1000)
                    try {
                        val isAlive = _connection.sendUnsignedV2(255u, 190u, heartbeat)
                        println("Is alive: $isAlive")

                    } catch (e: Exception) {
                        println("Exception: $e")
                    }
//                    val isAlive = _connection.sendUnsignedV2(255u, 190u, heartbeat)
//                    println("Is alive: $isAlive")
                }
            }
//
//            // Read  heartbeats in a new coroutine
            launch {
                _connection
                    .mavFrame
                    .filter { it.message is Heartbeat }
                    .collect {
                        val heartbeat = it.message as Heartbeat
                        println("System-Component ID: ${it.systemId} - ${it.componentId}")
                        println("Type/Autopilot: ${heartbeat.type} ${heartbeat.autopilot}")
                    }
            }
}
```

Result:

[image]

**UDP Server**

divyanshupundir, 2025-05-06:
@nghinguyengr Thank you for debugging this issue. You are probably correct. The UDP server in mavlink-kotlin is not exactly a true server that can connect to multiple nodes. It is mainly designed to do a handshake and work with only one client. Even the TCP server is designed that way.

If this does not fit your use case, you can implement and multiconnect UDP server in your local project. You can look at the existing UDP server in mavlink-kotlin.

You need to extend `AbstractMavConnection` and `UdpServerMavConnection` and implement the required methods. The rest of the library will work well with it.

divyanshupundir, 2025-05-06:
@nghinguyengr try launching the coroutines in `Dispatchers.IO`. Though it is handled by mavlink-kotlin, the frequent context switching might be causing interference.

nghinguyengr, 2025-05-06:
Hi @divyanshupundir , it's the same. 

Maybe the socket is fine. Let me debug further :(

[image]

[image]

nghinguyengr, 2025-05-06:
Hi @divyanshupundir  Finally, I found the difference between our heartbeat and QGC's heartbeat.

Our heartbeat:

[image]

QGC's heartbeat:

[image]

Do you think it affects?

nghinguyengr, 2025-05-06:
FYI: The heartbeat that I'm using is the same as the example. I just use connect.sendV1 to send it. The messages are the same right now. But the Fly Controller does not still receive my heartbeat.

[image]

divyanshupundir, 2025-05-06:
Oh. I have seen QGC and PX4 using MAVLink v1 for sending and receiving messages. If the flight controller is configured for MAVLink v2, then it can receive both types of messages. But if it is configured to MAVLink v1, then it won't be able to read v2 messages. Can you check the telemetry settings of PX4 on your flight controller?

divyanshupundir, 2025-05-06:
Also, you can try setting some other system and component IDs

nghinguyengr, 2025-05-07:
Hi @divyanshupundir  Sorry for the confusing info. On my remote control (MK15), QGC has the same problem with UDP, but it works well with Serial. 

Do you have a plan for supporting Serial connection?

divyanshupundir, 2025-05-07:
Hi @nghinguyengr. Please refer to #94 for serial connection. You can use [usb-serial-for-android](https://github.com/mik3y/usb-serial-for-android) to manage your serial ports. For that I have provided these implementations of [Sink](https://gist.github.com/divyanshupundir/4f191020db8f9d3b755f50ecb8ae35dc) and [Source](https://gist.github.com/divyanshupundir/52b1dbf95cb07bbc5a6d37568fc11370).

A few reasons for not adding Android serial support:
- mavlink-kotlin would require the Android SDK to build, which makes the library difficult to setup and manage
- Access to the serial port on Android requires somewhat complex handling of lifecycles and permissions, which is quite difficult to provide as a library and better handled in the application itself

nghinguyengr, 2025-05-07:
> You can use [usb-serial-for-android](https://github.com/mik3y/usb-serial-for-android) to manage your serial ports.

Yeah, I'm on my way.

nghinguyengr, 2025-05-07:
Honestly, I would like to contribute serial connection to this repo because I don't want to keep it for myself.

nghinguyengr, 2025-05-07:
> A few reasons for not adding Android serial support:
>
> * mavlink-kotlin would require the Android SDK to build, which makes the library difficult to setup and manage
> * Access to the serial port on [...]

First of all, I'll try it myself, then we will consider if it can be added to this repo.

nghinguyengr, 2025-05-08:
Hi @divyanshupundir, It works well. But we cannot add this connection to this repo, unfortunately.

divyanshupundir, 2025-05-14:
Yes @nghinguyengr. Integrating the flow into an Android project is quite difficult, considering the configuration challenges and different workflows.

divyanshupundir, 2025-05-14:
I'm closing this, as I assume the issue has been solved.

closed 2025-05-14 by divyanshupundir